WorldFamousElectronics/PulseSensor_Amped_Arduino #71: Using 2 Pulse Sensors: 1 Pulse Sensor has a bigger reading than the other

Camerutttt, 2018-04-20:
Hi, 1 of my Pulse Sensor outputs a higher reading as compared to the other pulse sensor. Why is that so? Here is the graph from the Serial Plotter: https://imgur.com/UXpDaQv
I am currently using the 5V pin and then connected to a breadboard column. The pulse sensor is powered from that column. This applies to the ground as well. Should I use the digital pins instead to power the pulse sensor like how it is done in the example?

biomurph, 2018-04-20:
Where are you placing the sensors? Fingertips? If you swap the position of the Pulse Sensors, do you get the same result? You are separating the two readings with <space comma space> Try just using a space, although I doubt that will change anything.

Can you send a photo of your Pulse Sensors? I want to verify they are made by us. There are many clones out there.

Camerutttt, 2018-04-23:
@biomurph

>  Where are you placing the sensors? Fingertips?

Yes, I just rest my fingertips on both sensors (Not using the velcro strap)

> If you swap the position of the Pulse Sensors, do you get the same result?

Originally sensor1 is connected to A0 and sensor2 is connected to A1. After swapping the analog pins (sensor1 -> A1 and sensor2 -> A0), the 2nd sensor still read a smaller reading as compared to the 1st. This is the result: https://imgur.com/a/g7HMe3N

> You are separating the two readings with Try just using a space, although I doubt that will change anything.

Tried it and it didnt work. 

> Can you send a photo of your Pulse Sensors? I want to verify they are made by us. There are many clones out there.

Here you go: https://imgur.com/a/BeXsvub

Camerutttt, 2018-04-24:
@biomurph tried swapping hands and its the same. It is always the 1st sensor which gives a bigger reading, never the 2nd sensor.

Camerutttt, 2018-05-09:
@biomurph 
Is it possible to increase the amplitude somehow? What if I connect the pulse sensor's 5V and GND pin to an external power supply instead of using the arduino's 5V and GND ports?

Camerutttt, 2018-05-09:
@biomurph When you mentioned playing around with the resistor, do you mean I should replace it with a SMD resistor of another value?

biomurph, 2018-05-09:
That's what I mean. Is that doable for you?

Camerutttt, 2018-05-11:
@biomurph Also, not sure if this is a viable solution, but what if I take the weaker sensor's reading and multiply it by a constant, will it actually help in making it easier to identify the threshold?

biomurph, 2018-05-11:
The R1 resistors are different because the LEDs came from different batches. 332 is 3.3K and 222 is 2.2K. if you have the capability, you could rework the R1 on the sensor that is lower signal amplitude.

Yes, you could multiply the lower signal by 2 or 3 to increase amplitude do do that :

First subtract 512 from the raw value. Multiply by n, then add 512 back to the result.
The signal is centered around 1/2 the Voltage.

Camerutttt, 2018-05-11:
@biomurph Alright, so I guess I will have to replace R1 with a lower value to increase the intensity of the LED? I will also try the mathematical method. Basically, I need the pulse to have a big enough amplitude to accurately determine the threshold, since I am trying to find the BPM of 2 people within a timespan of 10s using this method:  

 1) Find the time when each "heartbeat" occur in 10s (When the reading crosses the threshold) and store into an array
  2) Find the Interbeat Intervals (Time difference between each "heartbeat") and store into another array
  3) Find the median value from the Interbeat Intervals array
  4) Filter the Interbeat Intervals array by comparing it with the median.
   If a value in the Interbeat Interval array is within 80% of the median, add it into a 3rd array
  5) From the 3rd array, find the average
  6) Find the BPM, BPM = 60000/average (60000 since we are measuring within a timespan of 10s)

Not sure if this is a good method. Also, I am wondering does the amplitude of the pulse differ from person to person? So even if I use a very good sensor (getting high amplitudes for each pulse which allows me to determine a threshold easily), I may still get a weak amplitude when I am measuring a different person?

biomurph, 2018-05-16:
What you want to do sounds doable, for sure.

You will find that different people will have different amplitudes as well. This has to do with circulation, melanin, etc. Try it with math first.

Camerutttt, 2018-05-17:
@biomurph Yeah, since the amplitude are different, I cannot be using the same threshold for different people. How can I find a suitable threshold for each person without manually adjusting it in the code?

biomurph, 2018-05-17:
Check out this doc
https://github.com/WorldFamousElectronics/PulseSensorPlayground/blob/master/resources/PulseSensor%20Playground%20Tools.md
It lists the tools you can use in your sketch. There is a function call getPulseAmplitude() which returns the difference between the peak and trough of the most recent pulse wave. You can use that to 'tune' the threshold in real time. Note that the Pulse Sensor signal is centered around V/2, or 512 on the ADC (hence my recommendation to do the math if you want to 'amplify' your low signal). By watching the live beats for a while, you can get a good sense of how to use the amplitude value to tune a threshold. Something as simple as 
((amplitude / 4)*3)+512
might get you in a good spot to be above noise.

Also, you can use the amplitude value to ignore small signals that might be noise...

Camerutttt, 2018-05-17:
@biomurph Alright thanks a lot. I will look into it. What about an oximeter looking design to house the pulse sensor so that everyone exerts a constant pressure? Since the documentation mentions that "goldilocks" pressure must be exerted, this will allow everyone to exert the same amount of pressure. However, I'm not sure whether the amount of pressure exerted by each person, should be different due to the difference in their blood circulation etc.

biomurph, 2018-05-17:
The clip you are talking about is an interesting idea, but it is a different issue. Please bring it up on it's own separate isse thread.

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g