WorldFamousElectronics/PulseSensor_Amped_Arduino #72: Using an oximeter design for constant pressure?

Camerutttt, 2018-05-18:
I was wondering if I should design an oximeter to mount the pulse sensor and also to allow for people to exert a constant "goldilock" pressure so that the readings will be accurate.

Since the documentation mentions that "goldilock" pressure must be exerted by the user, what if everyone exerts the same amount of pressure with the help of the oximeter casing. Will that help in ensuring that everyone produces "goldilock" pressure? Or will it not work, since everyone's "goldilock" pressure is different, due to their blood circulation etc.

biomurph, 2018-05-18:
I think this is a great idea. We don't have any designs or plans for one in-house yet. I imagine that a 3D printable design would be useful for most folks. Do you have any ideas or sketches to share?

Camerutttt, 2018-05-19:
I took the Inventor file from here: https://www.thingiverse.com/thing:2052975/#files, but have edited it to fit the pulse sensor at the fingertip area. I have also made a hole for the wires to exit out of the oximeter. 

So you are saying that if everyone applies n Pa of pressure, whereby n is the amount for the "goldilock" pressure, the reading will be good?

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g