WorldFamousElectronics/PulseSensor_Amped_Arduino #79: problem with pulse sensor

nanoR4K, 2019-01-30:
so, im trying to build a glove with a pulse sensor and a OLED display, but im having a problem, the bpm value dosen´t show on the display instead it  aways shows 150-237bpm with and without my finger
(sometimes it just show 0)

im using an SH1106 1,3" 7 pin spi Oled display (the reset is on 5v if this helps)

ps: i dont have any idea where the problem is and im new at C

`#include <U8glib.h>
#define USE_ARDUINO_INTERRUPTS true
#include <PulseSensorPlayground.h>

const int PulseWire = A0;
const int LED13 = 13;
const int Threshold =570;
const int pinSCK = 4;
const int pinMOSI = 5;
const int pinCS = 6;
const int pinDC = 7;

int bpm = 0;

PulseSensorPlayground pulseSensor;

U8GLIB_SH1106_128X64 SH1106_Display( 4, 5, 6, 7 );

void DrawHeartBeat( int pulseBPM )
{
  char
  szBpm[10];

  SH1106_Display.setFont ( u8g_font_courB12);
  SH1106_Display.drawStr (8, 15 , " MG-MARK-1 ");
  SH1106_Display.setFont ( u8g_font_courB12);
  SH1106_Display.drawStr (3,54 , "BPM:");
  SH1106_Display.drawFrame(0, 0, 128, 64);
  SH1106_Display.drawFrame(72 ,41, 50, 16);
  sprintf( szBpm, "%d", pulseBPM );
  SH1106_Display.drawStr( 40,54, szBpm );
  SH1106_Display.drawBox( 74 ,44, 14, 10);
  SH1106_Display.drawBox( 90,44, 14, 10);
  SH1106_Display.drawBox( 106 ,44, 14, 10);

}

void setup(void)
{
  SH1106_Display.setFont(u8g_font_courB12);
  Serial.begin(9600);

  pulseSensor.analogInput(PulseWire);
  pulseSensor.blinkOnPulse(LED13);
  pulseSensor.setThreshold(Threshold);
  //
  if (pulseSensor.begin())
    Serial.println(" pulseSensor activated ");

}

void loop(void)
{
  UpdateBPM();
  UpdateDisplay();

}

void UpdateBPM( void )
{
  static unsigned long
  timeUpdate = 0;
  unsigned long
  timeNow;

  timeNow = millis();
  
  if ( (timeNow - timeUpdate) < 50 )
    return;
  timeUpdate = timeNow;

  bpm = UpdateHeartRate();
  if ( bpm >= Threshold )
    digitalWrite( LED13, HIGH );
  else
    digitalWrite( LED13, LOW );

}

void UpdateDisplay( void )
{
  static unsigned long
  timeUpdate = 0;
  unsigned long
  timeNow;

  timeNow = millis();

  if ( (timeNow - timeUpdate) < 250 )
    return;
  timeUpdate = timeNow;

  SH1106_Display.firstPage();
  do
  {
    DrawHeartBeat( bpm );
    Serial.print("BPM: ");
    Serial.println( bpm );

  } while ( SH1106_Display.nextPage() );

}

int UpdateHeartRate( void )
{
  return ( pulseSensor.getBeatsPerMinute() );

}`

biomurph, 2019-01-31:
There may be a collision between the pulse sensorcode and the OLED code. What Arduino are you using? Can you get the pulse sensor and the OLED to work on third own?

nanoR4K, 2019-02-01:
Im using an arduino nano and i don't know how to make this work

biomurph, 2019-02-04:
Can you get the pulse sensor and the OLED to work on their own?

closed 2021-08-05 by nanoR4K