WorldFamousElectronics/PulseSensor_Amped_Arduino #81: Issue with sensor voltage

HarrisL2, 2019-02-03:
Using the PulseSensor_BPM example, I was able to get the BPM effortlessly at first, however, after a few days I did the same thing again, but this time the values outputted by the program were nonsensical.
[image]
Here is the data from when the sensor stopped working properly. When it was working it looked exactly like those provided in the tutorials.

biomurph, 2019-02-04:
@HarrisL2 
You are posting this issue in the wrong repository.
https://github.com/WorldFamousElectronics/PulseSensorPlayground
In order to help you, I need to know what Arduino hardware you are using and also send a picture of your Pulse Sensor so that I can verify it is legit hardware.

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g