WorldFamousElectronics/PulseSensor_Amped_Arduino #85: pulse sensor cut off frequencies

holly1551, 2019-03-15:
Hi,
can someone please tell me the cut off frequencies of this sensor. I need it to obtain a respiration rate from the PPG signal. For this, I need the the frequencies to be atleast 0.2Hz and above.

biomurph, 2019-03-18:
@holly1551 The Pulse Sensor has some high pass filtration on it's output, so you will not see a classic PPG signal with DC offset and low frequency drift of the signal. 
If you want to derive respiration, you can use HRV.
https://github.com/WorldFamousElectronics/PulseSensor_Heart-Rate-Variabilty

closed 2019-03-18 by biomurph

holly1551, 2019-03-18:
Are you for certain that is doesn't contain frequencies from 0.2Hz and above? I cannot find how to derive respiration rate from HRV