WorldFamousElectronics/PulseSensor_Amped_Arduino #86: pulse sensor showing more than 200bpm

marklucass, 2019-05-13:
my sensor keep showing more than 200bpm. and sometimes eventhough i didnt put my finger the sensor keep counting.

my code : 

/*  Getting_BPM_to_Monitor prints the BPM to the Serial Monitor, using the least lines of code and PulseSensor Library.
 *  Tutorial Webpage: https://pulsesensor.com/pages/getting-advanced
 *
--------Use This Sketch To------------------------------------------
1) Displays user's live and changing BPM, Beats Per Minute, in Arduino's native Serial Monitor.
2) Print: "♥  A HeartBeat Happened !" when a beat is detected, live.
2) Learn about using a PulseSensor Library "Object".
4) Blinks LED on PIN 13 with user's Heartbeat.
--------------------------------------------------------------------*/

#define USE_ARDUINO_INTERRUPTS true    // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h>     // Includes the PulseSensorPlayground Library.   

//  Variables
const int PulseWire = 0;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
const int LED13 = 13;          // The on-board Arduino LED, close to PIN 13.
int Threshold = 550;           // Determine which Signal to "count as a beat" and which to ignore.
                               // Use the "Gettting Started Project" to fine-tune Threshold Value beyond default setting.
                               // Otherwise leave the default "550" value. 
                               
PulseSensorPlayground pulseSensor;  // Creates an instance of the PulseSensorPlayground object called "pulseSensor"

void setup() {   

  Serial.begin(9600);          // For Serial Monitor

  // Configure the PulseSensor object, by assigning our variables to it. 
  pulseSensor.analogInput(PulseWire);   
  pulseSensor.blinkOnPulse(LED13);       //auto-magically blink Arduino's LED with heartbeat.
  pulseSensor.setThreshold(Threshold);   

  // Double-check the "pulseSensor" object was created and "began" seeing a signal. 
   if (pulseSensor.begin()) {
    Serial.println("We created a pulseSensor Object !");  //This prints one time at Arduino power-up,  or on Arduino reset.  
  }
}

void loop() {

 int myBPM = pulseSensor.getBeatsPerMinute();  // Calls function on our pulseSensor object that returns BPM as an "int".
                                               // "myBPM" hold this BPM value now. 

if (pulseSensor.sawStartOfBeat()) {            // Constantly test to see if "a beat happened". 
 Serial.println("♥  A HeartBeat Happened ! "); // If test is "true", print a message "a heartbeat happened".
 Serial.print("BPM: ");                        // Print phrase "BPM: " 
 Serial.println(myBPM);                        // Print the value inside of myBPM. 
}

  delay(20);                    // considered best practice in a simple sketch.

}

based : https://pulsesensor.com/pages/getting-advanced
*i following all step by this site

marklucass, 2019-05-13:
im using arduino uno btw

biomurph, 2019-05-14:
@marklucass 
Sometimes you can get erroneous beats when the Pulse Sensor is not in touch with your body.
Pulse Sensor uses a light sensor that is sensitive to ambient light, so this could be part of the problem that you see.
Also, It could be your sensor. Can you please send a picture of the Pulse Sensor that you have, and your general setup?

marklucass, 2019-05-17:
sorry man kindda busy for couple days. btw here my sensor 
[image]

marklucass, 2019-05-17:
so i use laptop for powering the arduino, when my laptop connected adapter the sensor going more than 200bpm.  but when my laptop didnt connect to adapter my sensor didnt detect my finger at all.

biomurph, 2019-05-17:
@marklucass
It is well known that the Pulse Sensor can be perturbed by powerline noise when you are plugged in to a charger. We're working on a rev of the hardware to address this and other things.

Unfortunately, you have purchased a Pulse Sensor clone. Since I have no idea what is on that hardware, I can't help you troubleshoot.

Please support the original makers of Pulse Sensor by purchasing legit hardware. Here's a list of authorized sellers
www.pulsesensor.com
www.adafruit.com
www.sparkfun.com
www.amazon.com (lookout for clones! We are there, but so are they)
www.mouser.com
www.digikey.com

closed 2019-05-17 by biomurph