WorldFamousElectronics/PulseSensor_Amped_Arduino #90: BPM is giving way too high value

msaibhvani, 2020-03-03:
I am using the STM32f103c8 microcontroller using Arduino software with a pulse sensor and the value its showing is way too high it's going from 2000 to 4000. I am attaching my code here.
//  Variables
int PulseSensorPin = PA0; 
unsigned long int mybpm = 0; 
int i; 
int Sam_mybpm;
void setup() {
  Serial.begin(115200); 
}

void loop() {
  mybpm = 0;
  for (i=1;i<=1000;i++)
  {
    mybpm = mybpm+analogRead(PulseSensorPin);
    Serial.print(mybpm);
  }
  Sam_mybpm = mybpm/1000;
 Serial.print("Sam_mybpm"); 
  Serial.println(Sam_mybpm); 
  delay(1000);
    }
  The picture of my pulse sensor

[image]

msaibhvani, 2020-03-03:
The front side of the pulse sensor.
[image]

biomurph, 2020-03-03:
@msaibhvani 
First of all, the code you're using is not code that will determine BPM. I don't know what that code does, but it does not include our library or any of our other softwares.
It looks like you are simply reading the signal coming out of the sensor without doing any DSP on it.
Second, you are not using a legit Pulse Sensor. What you have is a clone. Since we don't know what is on tha that hardware, we can't troubleshoot it for you.

Please support the makers of Pulse Sensor and only purchase legitimate Original Pulse Sensor products.

sreenu926, 2021-08-17:
My pulse sensor is giving too high value(Around 1840). Please help me.

[image]
 
This is the sensor I'm using (Bought it from ROBU.IN):
[image]

**I'm using ESP32 DEVKIT Board.**

This is the program I'm using:

int PulseSensorPurplePin = 34;        // Pulse Sensor PURPLE WIRE connected to ANALOG PIN 0
int LED13 = 2;   //  The on-board Arduion LED

int Signal;                // holds the incoming raw data. Signal value can range from 0-1024
int Threshold = 1950;            // Determine which Signal to "count as a beat", and which to ingore. 

// The SetUp Function:
void setup() {
  pinMode(LED13,OUTPUT);         // pin that will blink to your heartbeat!
   Serial.begin(115200);         // Set's up Serial Communication at certain speed. 
   
}

// The Main Loop Function
void loop() {

  Signal = analogRead(PulseSensorPurplePin);  // Read the PulseSensor's value. 
                                              // Assign this value to the "Signal" variable.

   Serial.println(Signal);                    // Send the Signal value to Serial Plotter.

   
   if(Signal > Threshold){                          // If the signal is above "550", then "turn-on" Arduino's on-Board LED.  
     digitalWrite(LED13,HIGH);          
   } else {
     digitalWrite(LED13,LOW);                //  Else, the sigal must be below "550", so "turn-off" this LED.
   }

delay(10);
   
   
}

If my sensor is not legit, please suggest me a good pulse sensor to buy.

Have a blessed day,
Thank you.

biomurph, 2021-08-17:
@sreenu926 
Thanks for showing the pic of your sensor. You are not using a legit Pulse Sensor. What you have is a clone. Since we don't know what is on that hardware, we can't troubleshoot it for you.

Please support the makers of Pulse Sensor and only purchase legitimate Original Pulse Sensor products.

sreenu926, 2021-08-18:
Thank you so much Joel.
I told the same to ROBU.in guys and they apologized and promised me to send
new Legit sensor.
Thank you for helping people like me. May God bless you with his grace and
peace, AMEN.

yury-g, 2026-08-16:
Closing as resolved. The discussion identified that the sketch was averaging raw ADC readings rather than calculating BPM, and the hardware was not an Original PulseSensor. The requester confirmed the supplier would provide the correct sensor.

closed 2026-08-16 by yury-g