WorldFamousElectronics/PulseSensor_Amped_Arduino #92: High value input pulse sensor at 2000 without even touching it

johan-alda, 2021-05-26:
im using the example code from the playground and all output is 2k or more.

my laptop is charging while im running the code.

[image]

Both sensor that i use i bought it from 2 different place.

i also bought 3rd one and all still the same 2k or more.

i change my usb cable still same

the left jumper with word S i put it in A0 (esp32 wroom), middle cable + on 5v, right - on groung

595latch, 2021-05-26:
@johan-alda 
A couple of things need to be addressed.
First, this repo is superseded by the [PulseSensor Playground Library](https://github.com/WorldFamousElectronics/PulseSensorPlayground), Please use that library for best results and good data.
This repo is no longer supported.

Second, the Pulse Sensors that you have are not made by us. You purchased clones. I can't help you troubleshoot because I don't know what's on that hardware.

Please support the original makers of open-source hardware and purchase legit parts.
We sell on our site, as well as Adafruit, SparkFun, Mouser, Digikey.

I'm closing this issue. If you have other questions, please use the appropriate issue tab in a supported repositor.

closed 2021-05-26 by biomurph

johan-alda, 2021-05-26:
Ah I see thank you for your response