WorldFamousElectronics/PulseSensor_Amped_Arduino #95: Error in my Pulse Sensor Code

michellekulon, 2022-11-09:
Hello i need help with this code..
Have no idea why its the error

[image]

biomurph, 2022-11-09:
@michellekulon 
In order to help you troubleshoot, I need more information.
What Arduino board are you targeting?
What is your OS?
What sketch are you trying to program?

Also, this repository is deprecated. You should use the PulseSensor Playground Library.

yury-g, 2026-08-16:
Closing during issue cleanup because the information requested for troubleshooting was never provided and this repository has been superseded. For current Arduino projects, please use [PulseSensor Playground](https://github.com/WorldFamousElectronics/PulseSensorPlayground).

closed 2026-08-16 by yury-g